andrewshilliday/garage-door-controller #108: Feature Request: Use PiCamera to show live stream

zbowman, 2019-03-07:
Have been attempting to make this work on my own specifically for this project. Wanted to share it in case anyone else has a better means of implementing it.

Using code from : https://raw.githubusercontent.com/RuiSantosdotme/Random-Nerd-Tutorials/master/Projects/rpi_camera_surveillance_system.py

Attempting to put the camera view below the Left/Right buttons

Thanks again for making this garage controller. Have been using it for years and it's very reliable. Once the camera is working, will be making a 3D case that has a spot for the camera and possibly allows for rotating and angling the camera so each person can have the best garage coverage.

ronlh1, 2019-05-09:
See https://github.com/andrewshilliday/garage-door-controller/issues/68

commputethis, 2019-05-09:
I added one and also added a phone running IP camera.  However, my rpi died
a couple weeks ago so I can't access it.