andrewshilliday/garage-door-controller #110: automatic door closure after 10:30PM if a door is open for more than 30 minutes

Gilles94500, 2019-03-28:
We went home late last nigth and we forgot to close the gate. My mobile phone is on mute between 10PM and 8AM, so I have not been alerted and the gate has been left opened all the night.

Would be it possible to add an automatic doors closure if the time is greater than 10:30PM and a door has been opened for more than 30 minutes? And send a specific mail to know that it has happened.

It could be customizable using the json config file. Something like:
"use_autom":true
"autom_start":2230
"autom_duration":30

Gilles

rapi3, 2019-03-28:
you can add on /etc/crontab a simple bash or python script that will close/open the garage at any time you want...
#!/bin/sh
# Exports pin to userspace, change pin nr, sleep ...
echo "11" > /sys/class/gpio/export
# Sets pin 11 as an output
echo "out" > /sys/class/gpio/gpio11/direction
# Sets pin 11 to low
echo "0" > /sys/class/gpio/gpio11/value
sleep 1
# Sets pin 11 to high
echo "1" > /sys/class/gpio/gpio11/value
# UnExports pin from userspace
echo "11" > /sys/class/gpio/unexport
exit

python script:
import time
import RPi.GPIO as GPIO
GPIO.setwarnings(False)
GPIO.setmode(GPIO.BCM)
GPIO.setup(11, GPIO.OUT)
GPIO.output(11, False)
time.sleep(1)
GPIO.output(11, True)

zbowman, 2019-03-28:
If you already have your garage publicly accessible, this script might work for you. I'm using Pushbullet for notifications and had to use a few sleep commands. The pi doesn't respond fast enough if you ask for the status of each door one right after another but works ok with these delays.

`import requests
import time
from pushbullet import Pushbullet

pb = Pushbullet('<push bullet api token>')

right = requests.get('http://<controller IP or url>/st?id=right')
time.sleep(2)
left = requests.get('http://<controller IP or url>/st?id=left')

if right.text == 'open':
        pushright = pb.push_note("Garage Info", "Right Closing")
        closeright = requests.get('http://<controller IP or url>/clk?id=right')
        time.sleep(4)
elif left.text == 'open':
        pushleft = pb.push_note("Garage Info", "Left Closing")
        closeleft = requests.get('http://<controller IP or url>/clk?id=left')
        time.sleep(15)

rightcheckup = requests.get('http://<controller IP or url>/st?id=right')
time.sleep(2)
leftcheckup = requests.get('http://<controller IP or url>/st?id=left')

if rightcheckup.text == 'open':
        pushright = pb.push_note("Garage Info", "Right Closing")
        closeright = requests.get('http://<controller IP or url>/clk?id=right')
        time.sleep(15)
elif leftcheckup.text == 'open':
        pushleft = pb.push_note("Garage Info", "Left Closing")
        closeleft = requests.get('http://<controller IP or url>/clk?id=left')
        time.sleep(15)

rightfinal = requests.get('http://<controller IP or url>/st?id=right')
time.sleep(2)
leftfinal = requests.get('http://<controller IP or url>/st?id=left')

if rightfinal.text == 'open':
        pushright = pb.push_note("Garage Info", "Right Door STUCK")
        time.sleep(8)
elif leftfinal.text == 'open':
        pushleft = pb.push_note("Garage Info", "Left Door STUCK")
        closeleft = requests.get('http://<controller IP or url>/clk?id=left')

if (rightfinal.text == 'closed' and leftfinal.text == 'closed'):
        pushfinal = pb.push_note("Garage Info", "Both doors closed.")

`

I know my code is ugly, but it works.

Gilles94500, 2019-03-29:
Thanks; very nice scripts. I don't know the Pushbullet  pushbullet feature. I will give it a try. To be honest I am a little bit scary to be locked out in some cases. It is why I wanted to add this test (time and duration) .

Pack3tL0ss, 2019-04-10:
I forked and modified this project a 2-3 years ago to add something along these lines.
Slightly different, but adding a time of day bit wouldn't be difficult.

What I designed it to do:
- send notification if garage left open for 10 mins.  I added Pushbullet which has been added to master since

- send notification 60 seconds before autoclose
- audible warning 2 min before autoclose
- audible warning as autoclose becomes eminent
- retry loop that will give up after 5 attempts... Send notification on every failure... For times when an obstruction etc prevents for from closing
- autoclose enable/disable button in web interface
Added ZoneMinder integration and a lot of other stuff, unfortunately it's far deviated from the master branch at this point.  I have a lot of cleanup/ refactoring to do beforeI can post it, but I've been using it for a few years now.  It's a goal to get it up and start doing PRs, but it'll be a bit B4 I have time

Gilles94500, 2019-04-10:
Thanks for your help. Don’t worry. I am about to add some code to do the automatic closure. I also have modified the code to associate the garden lights along with a light detector, so not easy to share  with the community.

I was thinking to use a free  gpio that I could set to true or false to enable/disable the automatic closure.

Work in progress. I am not python expert :).

 

Gilles

andrewshilliday, 2020-01-28:
Hi all, I’m a bit late to the party on this issue, but just to throw in my own two cents.  I had considered adding a capability like that when I was originally writing the code but decided against it for the simple reason that the idea of garage doors opening and shutting on their own without any human involvement leaves me apprehensive.  

Safety issues aside, there is a security aspect to consider.  If for some reason your sensor should fail (loose wire, bad contact, etc.) the controller will think the door is open when it’s in fact close.  When 10:30 comes around, it will try to close it and instead will wind up opening the garage.  

I would suggest that a notification system would serve you better.