andrewshilliday/garage-door-controller #112: What about a Widget ?

Gilles94500, 2019-08-16:
Perdon my ignorance in scripting, but would it be possible to add a widget, something simple with one colored icon per door, the color to indicate the status, along with the status date/time ?
Thanks
Gilles

hadrins, 2019-11-02:
I simply saved the webpage to my home screen.   This question interest me though.   A quick look and I found an app in the play store called Meta Widget.   Will have to check that out. 

I have tasker,  Have been thinking in the past to make some triggers that will open the webpage if the phone is plugged in when it see the garage door pi device.   But don't want it activating because I just plugged it in to charge.

Gilles94500, 2019-11-02:
I already tried this app (Meta Widget) with  http://192.168.1.xx:8887/st?id=xxx but no way. It doesn’t show the expected answer (closed # 2019-11-02 16:39:39 ), so I can’t do anything.

hadrins, 2019-11-02:
I just tried it with just http://<ip address>/    with no password.   I don't recommend doing this as you can accidently open and close the door.    For me just having the webpage link saved to the home screen is good enough.