andrewshilliday/garage-door-controller #113: Door List Order

charredchar, 2019-12-30:
I added four doors to the config but they are not listed in order of the config or in order of the name. I want these to be listed in a specific order, how would I go about this?

andrewshilliday, 2020-01-28:
I hadn’t considered that.  It would require a code change to preserve the order from the config file.  Shouldn’t be too difficult if anyone wants to take a stab at it :).  Otherwise I’ll try to make the change when I get a chance.

bkjems, 2020-03-17:
Change 
`self.doors = [Door(n, c) for (n, c) in config['doors'].items()]`
to 
`self.doors = [Door(n, c) for (n, c) in sorted(config['doors'].items())]`

To sort by `doors` in the config file

andrewshilliday, 2020-06-18:
@bkjems would you care yo make a pull request for the change?

bkjems, 2020-06-18:
It looks like its committed.
https://github.com/andrewshilliday/garage-door-controller/commit/28557699b4ccf807b60a7e1b66558e9ce3c98233#diff-7a315c65deca2c033ce485bfa9de4f4a

nadineross412, 2023-03-03:
Hi there! In order to list the doors in a specific order, you may need to modify the code or configuration file of the garage door controller to reflect the order you want. You can try looking for a section in the code that lists the doors and try rearranging the order there. Alternatively, you can reach out to the developer or support team for the [garage door](https://besttechgadget.com/how-to-perform-garage-door-maintenance/) controller and see if they have any suggestions or solutions. Hope this helps!