andrewshilliday/garage-door-controller #131: Detect if door opener light is on

commputethis, 2020-12-27:
I have setup my garage door controller to add "doors" that turn on/off the lights on the garage door openers.  What type of sensor would I need to use to determine if the light on a particular opener is on or off?  I currently have the state_pin for the lights set to the pin for the opener it is on, so I have no idea if the light is on or off.  I have a camera in the garage and use this app to turn the light on to be able to see.

I hope that makes sense.

[image]

weishen412, 2023-02-13:
Hi there! That sounds like a great setup you have going on with your garage door controller. To determine if the light is on or off, you can use a photoresistor or a light-dependent resistor (LDR) sensor. These sensors work by detecting the amount of light in the environment and outputting a corresponding voltage or resistance value. By monitoring this value, you can determine if the light is on or off. I hope this helps and if you need any further assistance contact [Poquoson Quick Service](https://garagedoorrepairpoquoson.com/garage-door-replacement/) don't hesitate to reach out.

robertlope412, 2023-03-03:
Hi there! For your garage door setup, you may want to use a light sensor to determine if the light on a particular opener is on or off. This will allow you to accurately control the lights based on the state of the opener. You could also consider using a smart [Garage Door](https://techychimp.com/the-benefits-of-an-automated-garage-door/) light bulb that can be controlled through your app, which would eliminate the need for a separate sensor. Good luck with your project!