andrewshilliday/garage-door-controller #134: Python 3

eddiek2000, 2022-09-03:
Anyone done any work porting this project to python3?  I tried and updated a few things, but am not getting far.

commputethis, 2022-09-04:
I tried, but never could get it to work. So I find another way using MQTT and Home Assistant based on https://renjithn.com/garagepi-garage-opener-using-raspberry-pi/. I find this tutorial to not be quite right, so I added some documentation about what I did to get it to work. 

https://commputethis.github.io/MQTT-GaragePi/

eddiek2000, 2022-09-04:
Thanks.  I am still going to keep trying at the Python3 conversion.  I have used this app for years and really like it.  I have ZERO python/twisted/web experience, but am half way through a python course.  With that being said, I have just been able to get the doors and uptime to be displayed on the webpage.  Damn strings and byte strings/Python2/3.  Not sure if what I have coded is good, but seems to work.  Now I am working on the opening/closing/push control.  There is some difference between P2 and P3 with  'door = request.args['id'][0]' , but I'll figure to out.

I have a Buster install and a Bullseye install and am comparing the handling of messages/variables.  Hoping to have another breakthrough soon.  Found that http://garagepiserver/cfg and http://garagepiserver/upt were good debugging sources for passed messages.

eddiek2000, 2022-09-09:
Got it working with Python3 and python3-twisted.  Haven't tested anything other than the webpage being displayed correctly and the doors opening and closing and being displayed properly.  I don't use authentication or any of the alerts.  If anyone wants a copy, let me know.

Gilles94500, 2022-12-19:
Hi Eddiek2000,
My rpi2 has burnt, so I try to install the garage-door-controller on a rpi3B. Now, only python3 is available. I have used the 2to3 script to migrate to python3, then corrected some errors with zope.interface , then with PutChild statements. Now I have no more errors in the logs, but the webpage remains empty except the html header. I have tried to use the chrome inspect  tool and it seems that there may be an issue with JS.
Any idea ? Did I miss somùething ? You mention bove that it has worked for you without any security.
Thanks
Gilles

Gilles94500, 2023-02-13:
I give up and will try to rewrite it with flask instead of twisted.

eddiek2000, 2023-02-13:
I am so sorry for not seeing this earlier.

I am at work now,  but when I get home I can send over my edits/version.

As I posted look at the cfg and put pages on your garagepi.

Ed

Gilles94500, 2023-02-13:
Thanks Eddy. I am impatient hearing from you.
Gilles

eddiek2000, 2023-02-13:
As promised, my version:

[garage-door-controller.tar.gz](https://github.com/andrewshilliday/garage-door-controller/files/10727257/garage-door-controller.tar.gz)

Gilles94500, 2023-02-14:
Thanks a lot.

eddiek2000, 2023-02-15:
As I stated before, I don't have a cert for https or use auth, but I did find a bad line at Line 321.  I left out a b before clk.  The line should be:
root.putChild(b'clk', protected_resource)

This brings up the username password prompt, but they aren't accepted.  It is probably due to the way twisted is sending the user/pass back to python for authentication.  Python2/3 handle strings differently.  I am by no means programmer, but will try to look into this when I have a little free time.

Gilles94500, 2023-02-15:
I had already added the missing 'b' in the version I am working on.I have already worked on the migration.  and found some  (encode utf-8) to add . There is also a "//" to use instead of a "/" in (elapsed_time: value = seconds // length)
No way to make it work.
The way to migrate from twisted2 to twisted3 is described on the twisted site but they only talk abour the @implementer. 
I have not found an example which could be reusable in the controller.py. :(

eddiek2000, 2023-02-15:
> I found the error that I did with the twisted authentication.
> The password has to be in Byte with python3.
> args={self.config['site']['username']:self.config['site']['password']}
> A way to do it is : string = bytes(string, encoding= 'utf-8')
> You also need to encode the "ok" answer in the render with : return
> "ok".encode('utf-8')
>
> Now it works fine with my test raspberry which has no cabling yet.

Glad you found it.   I was pretty sure it was a byte issue.

NCAvian, 2023-11-22:
> I found the error that I did with the twisted authentication. The password has to be in Byte with python3. args={self.config['site']['username']:self.config['site']['password']} A way to do it is : [...]

Gilles, how did you fix these? Could you post the code?

Gilles94500, 2023-11-22:
I have tried to extract what has changed with Python3 along with twisted.

My problem is that I don’t remember exactly and I have amended the code to fit my home configuration.

I have found an old raspberry with python 2.7, so today I don’t use python3 until it crashes.

Hope it helps you anyway.

 

Andrew doesn’t answer anymore ☹

Thanks to him anyway

Gilles

 

See below:

 

from twisted.cred import checkers, portal

from twisted.cred.portal import IRealm, Portal

from twisted.web.guard import HTTPAuthSessionWrapper, DigestCredentialFactory, BasicCredentialFactory

 

 

 

        root.putChild(b'st', StatusHandler(self))

        root.putChild(b'upd', self.updateHandler)

        root.putChild(b'cfg', ConfigHandler(self))

IGNORE BELOW

        root.putChild(b'Uptime', uptime)

        root.putChild(b'au', gilles(self))

        root.putChild(b'noal', noal(self))

        root.putChild(b'qal', qal(self))

STOP IGNORING    

def run(self):

        task.LoopingCall(self.status_check).start(0.5)

        root = File('www')

        root

        if self.config['config']['use_auth']:

            clk = ClickHandler(self)

            arg1 = self.config['site']['username']

            arg2 = self.config['site']['password']

            arg3 = bytes(arg2, encoding= 'utf-8')

            args = {arg1:arg3}

            checker = checkers.InMemoryUsernamePasswordDatabaseDontUse(**args)

            realm = HttpPasswordRealm(clk)

            p = portal.Portal(realm, [checker])

            credentialFactory = BasicCredentialFactory("Garage Door Controller")

            protected_resource = HTTPAuthSessionWrapper(p, [credentialFactory])

            root.putChild(b'clk', protected_resource)

        else:

            root.putChild(b'clk', ClickHandler(self))

        site = server.Site(root)

        if not self.get_config_with_default(self.config['config'], 'use_https', False):

            reactor.listenTCP(self.config['site']['port'], site)  # @UndefinedVariable

            reactor.run()  # @UndefinedVariable

        else:

            sslContext = ssl.DefaultOpenSSLContextFactory(self.config['site']['ssl_key'], self.config['site']['ssl_cert'])

            reactor.listenSSL(self.config['site']['port_secure'], site, sslContext)  # @UndefinedVariable

            reactor.run()  # @UndefinedVariable

 

 

And you have to add “encode/decode utf8 ” to a lot of statements: ignore some which are specific to the code I use.

            arg3 = bytes(arg2, encoding= 'utf-8')

        door = request.args[b'id'][0].decode('utf8')

        return 'OK'.encode('utf-8')

        door = request.args[b'id'][0].decode('utf8')

        return "ok".encode('utf-8')

                            for d in self.controller.doors]).encode('utf-8')

           return "Alarms Deactivated".encode('utf-8')

           return "Alarms activated".encode('utf-8')

           return "Alarms are activated".encode('utf-8')

           return "Alarms are deactivated".encode('utf-8')

                return toto.encode('utf-8')

                return totoauto.encode('utf-8')

            return self.format_updates(request, updates).encode('utf8')

 

Good luck.

 

Cet e-mail a été vérifié par le logiciel antivirus d'Avast.
www.avast.com

NCAvian, 2023-11-22:
Thanks, Gilles! Everything seems to work except for authentication and SSL, although I haven't dug deeply into them yet. I've been tinkering with my code that puts the open/close times into the web page. With the latest version of Raspbian, they're now using journalctl instead of the log files. Cheers!

andrewshilliday, 2023-11-26:
> Andrew doesn’t answer anymore ☹ Thanks to him anyway

I'm sorry I haven't been responsive.  I don't really use the device anymore so I haven't been keeping up with this project.  If anyone else would like to officially take it over, I'd be happy to figure out how to transfer it.  And maybe if I get some time this week I'll work on updating the code.  It's been quite some time.

Gilles94500, 2023-11-27:
Unfortunately, I don’t have your skill 😊. I was a developer in the 80’s but Perl and Python were not yet born. Now I have problems with these modern languages because just to understand your code, I need, at least, to learn Python, Html, Css, JS and Ajax.

 

It would be very nice if you could adapt your py script to work with Python3 and the new twisted version.

I think that I made it work (see previous answers), but it is not as elegant as your script which is very logic and structured.

 

10 years after, I am still using it 😊

 

Thanks for that

Gilles